kiibohd/configurator #22: Firmware download sometimes fails, the first time and needs to be reattempted

jbondeson, 2017-10-14:
Could be a timeout or race condition...

jbondeson, 2017-11-08:
Reproduction has happened in prod-build

brokentone, 2017-12-06:
I can't seem to get a successful download, seems like this could help me out.

jbondeson, 2017-12-06:
@brokentone do you get a button that says "View Log" ?

brokentone, 2017-12-07:
I do, yes. Forgive me, I haven't read through enough of the error messages to know if any of it is sensitive and just post the whole thing, but here are the first few things that look to be going awry in the log:

```
-- Found Ctags: /usr/bin/ctags (found version "5.9~svn20110310") 
ctags: cannot open tag file : Permission denied
...
-- Build files have been written to: /srv/KiiConf/tmp/3eb1cf0cb19ecd9f6fe5d4ad537d2ee315a28661d469f2
[1/35] Generating KLL Layout
FAILED: cd /srv/KiiConf/tmp/3eb1cf0cb19ecd9f6fe5d4ad537d2ee315a28661d469f2 && /srv/KiiConf/controller/kll/kll --version && /srv/KiiConf/controller/kll/kll --kiibohd-debug --config /srv/KiiConf/controller/Scan/Devices/ISSILed/capabilities.kll /srv/KiiConf/controller/Scan/Devices/MatrixARMPeriodic/capabilities.kll /srv/KiiConf/controller/Scan/Devices/PortSwap/capabilities.kll /srv/KiiConf/controller/Scan/Devices/UARTConnect/capabilities.kll /srv/KiiConf/controller/Macro/PartialMap/capabilities.kll /srv/KiiConf/controller/Macro/PixelMap/capabilities.kll /srv/KiiConf/controller/Output/HID-IO/capabilities.kll /srv/KiiConf/controller/Output/pjrcUSB/capabilities.kll /srv/KiiConf/controller/Debug/latency/capabilities.kll --base /srv/KiiConf/controller/Scan/K-Type/scancode_map.kll --default /srv/KiiConf/controller/kll/layouts/stdFuncMap.kll KType-Standard-0.kll --partial /srv/KiiConf/controller/kll/layouts/stdFuncMap.kll KType-Standard-1.kll --emitter kiibohd --def-template /srv/KiiConf/controller/kll/templates/kiibohdDefs.h --map-template /srv/KiiConf/controller/kll/templates/kiibohdKeymap.h --pixel-template /srv/KiiConf/controller/kll/templates/kiibohdPixelmap.c --def-output kll_defs.h --map-output generatedKeymap.h --pixel-output generatedPixelmap.c --preprocessor-tmp-path /srv/KiiConf/tmp/3eb1cf0cb19ecd9f6fe5d4ad537d2ee315a28661d469f2/tmp_kll
...
[Token('ScanCodeStart', 'S['), Token('Number', '0x00'), Token('Dash', '-'), Token('Number', '0x5F'), Token('CodeEnd', ']'), Token('Operator', ':+'), Token('AnimationStart', 'A['), Token('Operator', ':'), Token('Name', 'fingerprints'), Token('CodeEnd', ']'), Token('Parenthesis', '('), Token('Name', 'start'), Token('Parenthesis', ')')]
got unexpected token: 1,4-1,4: Operator ':'
NoParseError got unexpected token: 1,4-1,4: Operator ':'
Bad kll expression, usually a syntax error.
```

This all happens trying to generate any config, including the default one.

jbondeson, 2017-12-09:
@brokentone I believe your issue was bug #70 I have fixed it in v0.4.1