kiibohd/configurator #35: bin file is absent from the downloaded zip when the keyboard layout has been changed

Iris3Meow, 2017-11-05:
When I change the keymap even a little, the downloaded zip file does not contain the bin file needed for the flash. 
The keymap change I am currently trying is changing the base PgUp key to Vol Up key and the PgDn key to Vol Dn key. 
I then assigned the Pg Up and Pg Dn functions on the respective buttons on Layer 1.

[image]

Iris3Meow, 2017-11-05:
This issue has not been fixed as of 0.2.0 I have tried just now
[image]

jbondeson, 2017-11-05:
@Iris3Meow compilation failed due to an error. in this case it's a blank animation in your file it's `white` if you were to delete it your compilation should succeed. 

The "success" message is incorrect and will be addressed by #25 and I opened #36 to address the root cause of this compilation failure.

closed 2017-11-05 by jbondeson

Iris3Meow, 2017-11-05:
@jbondeson 
So 2 questions.
1. how did you know i had a layer labeld white?
2. how else am I supposed to get a colorscheme where the keyboard has white light and does a fingerprint animation of a random color upon press? I don't see any option where the user can create LED animations

Thanks for answering my question btw!