kiibohd/configurator #36: Don't send blank animations to the server

jbondeson, 2017-11-05:
This causes compilation to fail.

Iris3Meow, 2017-11-05:
@jbondeson
So 2 questions.

how did you know i had a layer labeld white?
how else am I supposed to get a colorscheme where the keyboard has white light and does a fingerprint animation of a random color upon press? I don't see any option where the user can create LED animations
Thanks for answering my question btw!

jbondeson, 2017-11-05:
@Iris3Meow Your screenshot had the hash for the download so I just quick checked out the build.log file and the .JSON on the server

We don't have a random color in yet, but we have an open issue on the controller to support it in the future.

Iris3Meow, 2017-11-06:
@jbondeson 
I am very interested in making my own color schemes.
Is there a controller program you guys are using to make the pre-built animations (like miami waves)?
Or are you litearlly writing codes to program them?

If there is a controller program that does it for you, like Kiiohd does for the keymap, I would like to know where I can download it. You might be able to tell, but I mostly purchased the k type because I am interested in programming my own LED color schemes. It also looks insanely cool.

jbondeson, 2017-11-06:
@Iris3Meow those are a combo of hand written and modified by the configurator code. If you want to talk about it in depth join our discord https://discord.gg/9tpgDGS !

jbondeson, 2017-11-06:
Fixed on the server side with https://github.com/kiibohd/KiiConf/commit/3f329d2b04cef5d9ce080ae717589a099e01d3db

closed 2017-11-06 by jbondeson