kiibohd/configurator #37: Layer Reset Action

TheIronMarx, 2017-11-05:
When switching between layers, I sometimes prefer to use the NEXT and PREV actions to scroll through them. With the K-Type, there currently isn't any indication that informs me what layer my board is presently on. I would like to be able to map a button that resets what layer I'm on so if I can't remember, I can at least reset and start over.

jordandcarter, 2017-11-06:
I've got the ergodox but had a similar issue when wanting to go back to base layer. It wasn't entirely clear to me (because I didn't read the docs) but you can use the LOCK-1, LOCK-2, LOCK-3... to switch back to the base layer, so long as you are on the layer matching the lock.

From the KLL Spec
> Lock - Enables keymap until lock is pressed again

So, for each layer add a button which is the corresponding LOCK-#, then when on that layer, hitting that button should put you back to base layer.

I know you can stack layers, and if that is what you are doing then I think LOCK-# might only pop the current layer from the stack, and not return you all the way back to base. I never stack, so that's not been an issue for me. In fact the reason I wanted this was I'd mis-type and end up with stacked layers and be confused about how to unstack them all and get back to base.

In my setup I have the esc button as LOCK-1, LOCK-2, LOCK-3 for those layers, so I can always just mash the esc button if I end up on a layer I don't want to be on. I'm a vim user, so mashing esc comes naturally to me.

dmgm, 2017-11-06:
Would it be possible to change the colour of the number 1 to 7 to indicate layers that are enabled?
The layers on the configurator are colour coded, perhaps the same colours could be used?