kiibohd/configurator #39: White Balance Animations and Colours

Diserasta, 2017-11-06:
If you have 255,127,0 (orange), and you halve it, ideally you should have 127,63,0. This doesn't seem to be the case though, as setting your keyboard to 255,127,0 and turning brightness down results in the light quickly turning yellow and finally green before finally turning off.

To me, this indicates that the red LED is being cut away faster than green. I assume this is because the Red LED is brighter than the others (this tends to be the case with RGB LEDs due to the compounds used), but it seems like it's over-correcting. This could be the WS2812 itself doing some sort of white correction (I suspect it is since white looks fine). We may need to correct against this, as it's distorting red-primary colours. This may be as simple as scaling up red each time we step down brightness, or something more complex.

bradyjoh, 2017-11-09:
Not sure if this is helpful, but I've done a number of little projects with WS2812s and WS2801s using the FastLED library https://github.com/FastLED/FastLED and it has really wonderful options for color correction (https://github.com/FastLED/FastLED/wiki/FastLED-Color-Correction)

haata, 2017-11-09:
Some more info, the ISSI chip (LED Driver) on the K-Type has some information on gamma correction as well.
http://www.issi.com/WW/pdf/IS31FL3733.pdf

adrianlu95, 2019-11-14:
Is there any update on this? The issue still exists now for the Gemini version of the configurator, and the issue is now a couple years old.