kiibohd/configurator #45: When pressed in succession, some of the fingerprints do not go off.

Iris3Meow, 2017-11-08:
[KType-Standard-e1fecc954677943987b9ee9f5d000cd6.zip](https://github.com/kiibohd/configurator/files/1455353/KType-Standard-e1fecc954677943987b9ee9f5d000cd6.zip)

The one I attached has a two tone fingerprint warp (green to yellow).
When you press keys such as asdf in succession, the fingerprint does not go off on sdf.
Same can be seen with other successions like bnm,

bradyjoh, 2017-11-10:
I thought the two-tone fingerprint was intended to change to color 1 when pressed down and then leave color 2 on the key afterwards! Are you sure this isn't the intended behavior? I'm seeing this too, by the way.

nrivard, 2017-11-14:
This is also happening even on the simpler fingerprints. seems like fingerprints only register _after_ the previous animation has completed. so if your fingerprint triggered animation takes a little bit of time (like my custom one does) then no other fingerprints register until it has completed.

jbondeson, 2017-11-14:
I think this is a little more complicated. On the stock fingerprints if you type `asdf` quickly it will not trigger all four, however if you type `fdsa` it will. My guess is there's an animation trigger bug in the controller. Right now there's a bunch of work happening on that front so I'll keep watching this as I'm testing.

nrivard, 2017-11-14:
@jbondeson ha i see that on mine as well. interesting. i will wait for some fixes to see how it shakes out.

sidenote: is there a guide to the animation language? For fingerprints, i wanted to add a water ripple effect but adding the following didn't work:
```
# single frame only for illustration purposes. this, in my mind, should color the key to the left of the pressed key white, but it doesn't work.
P[r:i,c:i](248,9,154), P[r:i,c:i-1](255,255,255);
```