kiibohd/configurator #55: relative indexing of pixel values not working

nrivard, 2017-11-15:
According to the draft spec located at https://www.overleaf.com/read/cfqydmrpfdrs#/7429832/, lines 1274-9, I should be able to index "pixels" by relative indexing. However, the following doesn't work. I am trying to make a raindrop-like effect where a key press radiates out to surrounding pixels.

NOTE: i am only showing 3 frames for this though i have many more frames to see the actual effect.
In this example it should show white on the keypress in frame 1, then show white on the key, key left of key and key right of key, and on the last frame, show white on the key, the 2 keys left of key, and the 2 keys right of key.
```
P[r:i,c:i](255,255,255);
P[r:i,c:i](255,255,255), P[i:-1,c:i](255,255,255),  P[i:+1,c:i](255,255,255);
P[r:i,c:i](255,255,255), P[i:-1,c:i](255,255,255),  P[i:+1,c:i](255,255,255), P[i:-2,c:i](255,255,255),  P[i:+2,c:i](255,255,255);
```

calebadcock, 2017-11-28:
The specs show the following:

- `P[i]`: Relative pixel index
- `P[r:i]`: Relative row fill
- `P[c:i]`: Relative column fill
- `P[r:i,c:i]`: Relative rectangle index

Notice that pixel indexing is just a number; it does not use `r` and `c`.

nrivard, 2017-11-28:
Except the built-in animations do use keypress vars addressed by `r:i,c:i`. See the `fingerprint` built-in animation. It is a rectangle of `1` i assume...

But even so, how is the currently pressed key known via straight pixel addressing? It sounds like it is not a 2d array of pixels but rather a flat array that is physically laid out in a 2-dish config but in terms of addressing in code it is a 1d array?

tryptech, 2018-01-10:
It seems to be a 1D array. Currently, if you address a pixel that would go outside of columns 0 to 37, the next pixel that is addressed is in the row above or below. If this so happens to go beyond row 0, column 0 or row 7, column 37 the keyboard becomes unresponsive.