kiibohd/configurator #58: Ergodox Infinity LED's not working

supermitsuba, 2017-11-22:
Hello, I dont know if this is the right project or the compiler thing.  Here is what I am seeing,  I have kll based on 0.3.  The new configurator works fine uploading that.  

When I use a layout from the configurator and it generates a new kll bin based on 0.5, my leds dont work.  

I have tried to use LED toggle, i have compared my JSON, but I am stumped why the leds dont work anymore.

jbondeson, 2017-11-22:
This should be fixed shortly, code was committed this week to re-enable the LEDs to the controller project. There’s a few more tests and tweaks necessary on the configurator side but should be pushed later this week.

I’ll close the issue when it has been pushed.

Freddster, 2019-10-15:
@jbondeson
Just stumbled upon this issue when searching for something else. Do you know if the changes has pushed or is this still an issue?

pnhowe, 2020-08-24:
I'm using Configurator 1.1.0, and keyboard firmware v1222, and my backlights are not working.  Has this issue been fixed, or am I seeing something else?

aimersville, 2021-03-13:
I have the same issue, after flashing using Configurator 1.1.0, my LED backlights are gone. Adding LED On/Off/+/-/100% still has no effect.

nyan-lin-tun, 2022-02-14:
Have you double check that you choose the right shine or glow in [ORXY](https://configure.zsa.io/) Mine one fixed by that way.