kiibohd/configurator #63: "?" macro error

hallucinatesgirls, 2017-11-29:
The configurator is assigning "e?" instead of "?". Cheers

jbondeson, 2017-11-29:
Can you be a little more specific as to what is happening?

Also OS and the language setting will be helpful no doubt.

hallucinatesgirls, 2017-11-29:
There's a typo in the code where an e has accidentally been added. You can see this in the JSON raw layout:
[image]
OS: Windows, Language: English

hallucinatesgirls, 2017-11-29:
This appears to be the culprit in the predefined.cljs file
[image]

jbondeson, 2017-11-30:
What a fun typo. I'll get that into the next build later that I'll be pushing later this week.

commit 01ea66e1 (2017-11-30): Fix typo in '?' macro - Fixes #63
closed 2017-11-30 by jbondeson