kiibohd/configurator #74: Compilation fails very often on Mac

daveygm, 2017-12-19:
I'm using a K-Type with a MacBook Pro and constantly get failures when trying to compile in the Configurator. It seems as though changing anything to do with the visuals really causes problems. This is my most recent example where I tried to have to have 2 different visual sets (both single colors) and tried to compile.

Here's the build log:
[build.log](https://github.com/kiibohd/configurator/files/1573524/build.log)

### Configuration:
MacBook Pro (15-inch, Late 2016)
2.9 GHz Intel Core i7
macOS 10.13.2 (17C88)

harper357, 2018-01-24:
I am also having this problem.  I am using Version 0.4.1 (0.4.1.112), same computer as above.

I did some testing, with just the standard firmware it compiled fine. 
I add a custom static LED Map to the standard and the compile fails.
I add a preset animation to the standard and the compile is fine. 
I add a preset animation and a custom LED map to the standard and it compiles fine.

harper357, 2018-01-24:
OK, I played around with things and I think I have it figured out. 

It seems that if there is an empty Custom KLL line `;`, this crashes the compiler. 

I have no idea how/why an empty line sometimes shows up in the Custom KLL. It seems to sometimes just show up. I was playing around with adding and deleting animations (both preset and custom) and it would just sometimes show up.