kiibohd/configurator #80: Fingerprint animation for underglow

Iris3Meow, 2018-01-10:
I would really like to get an option for the fingerprint to work on the underglow as well.
For example, If using the single tone, I would like to be able to designate certain underglow RGB  to also activate the fingerprint animation upon a keystroke (i.e. when I press "G" both the G key and some of the underglow RGB I picked out glows in green).

sK0pe, 2018-01-28:
You can already do this.
Inside of the fingerprints animation in the "frames" key, add keys 88 - 119 with the green colour next to each of the frames that look like this:
`P[r:i,c:i](x,y,z)`
Where x, y and z are 0 - 255.

montydrei, 2018-04-11:
So, something like this?
https://pastebin.com/DcQXuxk2

montydrei, 2018-04-11:
So, something like this?
https://pastebin.com/g2tmwn6J