kiibohd/configurator #81: infinity 60% has bad bin file

fessmeier, 2018-01-13:
The FW compilation is broken for the infinity 60% keyboard.  The same is true with the web site.  It appears that the main layer is missing.  Layer 1 and 2 are working for my keymap.

thomasrebele, 2018-05-14:
I have the same issue. As I workaround I modified the json file, and shifted the layers (0->1, 1->2, 2->3, ...). Then I uploaded it to the [input.club configurator](https://input.club/configurator-ergodox/). The website shows a wrong layout (i.e., empty keys on the main layer), but the firmware works as expected.

jaredbidlow, 2020-01-29:
Does the input.club configurator still exist? I'm looking for a workaround as suggested by @thomasrebele, as the configurator I downloaded from their website, and use on my Windows laptop, is not altering the base layer. Is there a manual method to configuring this keyboard, the infinity 60%, in my case the alphabet layout.

thomasrebele, 2020-02-02:
@jaredbidlow, it seems that the input.club configurator has been removed in favor of a configurator that runs locally. Have you tried the workaround that I described?