kiibohd/configurator #82: configurator produces bad bin file for k-type

alexeiz, 2018-01-23:
I uploaded my json config to the web configurator and dowloaded the "latest" firmware.  Then flashed the bin file using the Windows desktop configurator.  Flashing succeeded, but the keyboard would not respond after that.  Although, it's still possible to put the keyboard into the flashing mode and flash the previous version of kiibohd.dfu.bin (which I luckily saved).  Same thing happens if I use the desktop configurator to obtain the bin file.  My previous version of the bin file which still works was created on Jan 23 2018.

alexeiz, 2018-01-23:
here's my layout json config file: https://pastebin.com/NG4eWsLR

InfinityTotality, 2018-01-23:
Importing that json into the desktop configurator actually works on my k-type. I have noticed that there seems to be a limit on how big a config can be though. If you delete one of those animations, it may work for you.

alexeiz, 2018-01-24:
I can import json without a problem.  The problem is with the bin file  generated for that configuration.  When I flash that bin file to the keyboard, the keyboard doesn't work: no keys register.

InfinityTotality, 2018-01-24:
Yes, I mean that I am able to compile a bin file, flash it to my k-type, and have it boot and function using your configuration.

james-cosgrove, 2020-04-01:
I'm also having the same issue on the desktop configurator for my Infinity 60%. The .bin file I currently have that was made using the  old web configurator still works, but if I create a new layout and flash it, none of the keys work.

haata, 2020-04-01:
@james-cosgrove which Infinity 60% do you have? Did it have optional LEDs?