kiibohd/configurator #83: K-Type Will Not Wake Mac From Sleep

tthurston4, 2018-01-26:
The initial firmware that shipped with the K-Type used to wake the Mac from sleep. Now, after using the Kiibohd Configurator v0.4.1 and flashing the K-Type with updated firmware, waking from sleep with a keypress is no longer possible.

sK0pe, 2018-01-28:
What operating system, what Mac, what USB port?

tthurston4, 2018-01-28:
OSX 10.10.5, Mac Pro Mid 2012, USB 2.0 port in the back of the tower.

rafaelgonzalez, 2018-01-30:
Same here with:

OSX 10.13.2
MacBook Pro (Retina, 15-inch, Mid 2015)

BinaryHammer, 2019-08-08:
Tried it with macOS 10.14.6 and the new 1.0.1 configurator. MacBook Pro 15" 2017 connected with USB-C to USB-C cable with the K-Type. MacBook in sleep, pressing a key will wake up the MacBook.