kiibohd/configurator #84: Download Firmware failing 100% of the time

BrianJFeldman, 2018-01-29:
Attempting to compile and download firmware from Kiibohd Configurator v0.4.1.112 on mid 2015 MacBook Pro on 10.12.6 fails 100% of the time. 

Build.log here: 
[build.log](https://github.com/kiibohd/configurator/files/1675222/build.log)

It looks like a stray colon in the kll? Help appreciated.

InfinityTotality, 2018-01-29:
The log seems to imply you have a line like this in custom KLL:
S[0x00-0x5F] :+ A[:fingerprints](start);

If that is there, the colon right before fingerprints is extraneous and causing this error. Try removing it. It should be in custom KLL for your base layer.

BrianJFeldman, 2018-01-30:
looks lke you were right. Edited it out and started working thanks!

closed 2018-01-30 by BrianJFeldman