kiibohd/configurator #89: Mouse up/down keys reversed (both x1 and x5)

ToyKeeper, 2018-04-01:
Using kiibohd-configurator-0.4.1-linux-amd64.deb on my brand new whitefox vera, it appears that the mappings for Mouse Up and Mouse Down are swapped.  This affects both the x1 and x5 speeds.

For a layout, it's easy enough to just swap them to make the keys work as intended, but the configurator should probably have the right mappings by default.

ToyKeeper, 2018-04-01:
The json it creates comes out like this.  Looks like it just needs to swap 1 and -1 in the output, or swap the up/down glyph.
```
{
    "key": "#:mouseOut( 0, 0, -1 )",
    "label": "↓ x1"
}
{
    "key": "#:mouseOut( 0, 0, 1 )",
    "label": "↑ x1"
}
```

dmgm, 2018-05-15:
This also affect the k-type under macOS. v4.1 of the configurator.

dmgm, 2018-09-14:
Is there someone this could be assigned to?

Phiphel, 2019-09-07:
This is still bugged.