kiibohd/configurator #90: 'click to open' link lacks context menu or ability to see the link

ToyKeeper, 2018-04-10:
When I click "Download firmware" and then "click to open", the configurator spawns a music player with irrelevant files loaded into a playlist.  Like, ringtones I made a decade ago.

It would be nice to be able to see what the link is, perhaps by right-clicking and selecting "Copy link address" like in a normal browser window, but the context menu seems to be disabled.  The link details which would normally show at the bottom of the window are also missing.

Could something be added to allow the user to see and/or copy the link without relying on the host OS's launcher mappings being correct?

jbondeson, 2018-04-10:
@ToyKeeper what OS? This actually uses the electron [show item in folder](https://github.com/electron/electron/blob/master/docs/api/shell.md#shellshowiteminfolderfullpath) api to view the .zip file.

ToyKeeper, 2018-04-10:
It's Debian, but it's weird.  It's not using a common desktop environment like GNOME, and it intentionally lacks anything resembling a file browser program or a GUI zip viewer.  The electron shell module's description is "Manage files and URLs using their default applications", but the default-application mechanism doesn't really exist on my system.

It'd be nice to have some sort of fallback when the host OS isn't configured as expected.  Ideally, showing or copying the path somehow, so it can be used in a terminal.