kiibohd/configurator #95: Inf. Ergodox: flashing 'bin' generated by `configurator' breaks LOCK keys

srustamo, 2018-05-01:
Version: kiibohd-configurator-0.4.1-win
OS: WIN7 64b

Flashes without error. But LOCK keys for layers toggle only 'ON' and consecutive depresses on them do not deactivate the locked layer.

Creating a layout from the web configurator, selecting 'Latest' option also breaks the LOCK key as above. 

Selecting 'lts' in the web configurator produces working bin files.

srustamo, 2018-05-01:
> Also a quick work around is simply to add the appropriate lock key on it's own layer (i.e. add LOCK-2 to layer 2)

Does this mean if I want to have a LOCK-2 key on Main layer, I need to add LOCK-2 key on Layer 2 on the same key position?

> I was only seeing this on the right hand (LOCK keys on my left deactivated normally) for my ergodox, did you observe it on both?

Wound up the project for today, will check tomorrow.

jbondeson, 2018-05-01:
> Does this mean if I want to have a LOCK-2 key on Main layer, I need to add LOCK-2 key on Layer 2 on the same key position?

That is an easy workaround for the issue, yes.

srustamo, 2018-05-02:
Having LOCK-2 on the the second layer works. Thanks. 

As to broken LOCK keys on the right side only: indeed only right side is problematic.

closed 2018-05-02 by srustamo