kiibohd/configurator #98: Writing Macro

munnybear, 2018-05-29:
Hi, 

I know in the configurator interface, the "Macros" section currently sais "coming soon ಠ_ಠ"

However,  I am wondering if it is possible to do this via json, then importing it into the configurator?

If so, is there any document that shows how key combination should be written?  I've had a go using the concept from KLL, by using "+" and "," with no luck...

Thanks in advance!

munnybear, 2018-05-29:
@scrathe that's the KLL language. May I ask how did you put the KLL script onto the keyboard/bin file? I tried changing the .kll file for each layer with 0 luck........ did you use the configurator?

scrathe, 2018-05-29:
Yeah I used the configurator. I'm very new to the keeb scene and still reliant on GUI.

munnybear, 2018-05-30:
@scrathe can you please shed some light on how you managed to register the KLL script using the configurator. Right now, when I type something under "Custom KLL"... nothing happens :S

closed 2019-01-10 by jbondeson