kiibohd/configurator #102: adding canned animation creates invalid custom kll

keybeejeebee, 2019-01-03:
- starting with a blank layout (standard)
- click on show visuals
- click customize canned layout
- select fingerprint two toned
- click create
- click hide visuals
- click custom kll.

This adds the required custom kll for the reactive animation, however it contains the invalid keyword undefined:
```
undefined
#~~~START ANIMATION (fingerprints_two_tone) INJECTION~~~#
S[0x00-0x5F] :+ A[fingerprints_two_tone](start);
#~~~END ANIMATION (fingerprints_two_tone) INJECTION~~~#
```

from line 109 in customized-canned.js
`updateCustomKll(customKll['0'] + addition, 0);`

I need to figure out how to do pull requests, but for now a temporary fix (looks like you have plans for this area) is something like:
```
      if(customKll[0] && customKll[0].length) {
        updateCustomKll(customKll['0'] + addition, 0);
      } else {
        updateCustomKll(addition, 0);
      }
```

keybeejeebee, 2019-01-03:
I'm sure you can tell from my notes, but I should have specified this is configurator-next

jbondeson, 2019-01-03:
@keybeejeebee thanks for the catch, was testing in Kira for too long where it had custom KLL by default!

commit c8f97bc2 (2019-12-03): Properly default default layer custom kll when injecting animation kll

Fixes #102