kiibohd/configurator #104: Show Log button does nothing

jrwrigh, 2019-01-08:
_From @jrwrigh on January 8, 2019 19:5_

* **Device**:
Infinity Ergodox, both hands plugged in and in use

* **Firmware Origin**:
N/A

* **OS**:
Windows 10 10.0.17134 Build 17134

* **Version**:
N/A (?)

* **Reproduction Steps**:
1) Go to Infinity Ergodox page of the configurator
2) Load (seemingly) any layout
3) Hit "Download Firmware"
4) "Compilation Failed" notification pops up

* **Resulting Issue**:
The "SHOW LOG" button of the notification does nothing.

[image]

I'll create another issue for the actual compilation fail part and link it here. For now, the "SHOW LOG" button doesn't help.

_Copied from original issue: kiibohd/controller#304_

jrwrigh, 2019-01-08:
Note this is in v0.6.0 alpha. I didn't realize that I had installed an alpha version. This would've been helpful information. 

I already created #305 for addressing the difficulty of seeing what version of the configurator I'm running.

jbondeson, 2019-01-10:
What this means is there actually is no log, probably an HTTP error, I'll update the popup to show the actual error in this case.

commit f5eba919 (2019-03-06): Don't offer to show log if there is none

Fixes #104
closed 2019-07-10 by jbondeson