kiibohd/configurator #110: Unable to find keyboard in flash mode (Whitefox) when trying to flash firmware

Qwertee, 2019-01-15:
First off, I apologize if this is the incorrect repo to submit this issue.

OS: Windows 10 64 bit - 1803
Configurator version - 1.0.0
Tried on 2 different Windows 10 laptops

I recently acquired a Whitefox keyboard and I am trying to use the configurator to flash new firmware to it, but it cannot seem to recognize that my keyboard is in flash mode. I am able to download the firmware and get a "compilation successful" message, but the flash screen says "No Keyboard Found in flash mode. Press the reset button on the bottom of your keyboard to enter flash mode." which I have tried multiple times across multiple computers. I may be missing something obvious here, but I tried searching relevant issues and looking through the documentation to no avail.

Thanks!

RikerCabra, 2019-01-23:
Did you install the driver with zadig as mentioned [here](https://input.club/configurator-setup/)? I was having the same issue on a new PC and forgot to do that step, but following that article fixed my problem.

ChrisMcPherson, 2019-02-05:
Verifying and fixing the drivers in both normal and flash mode fixed this same issue for me with the new KIRA keyboard. Thanks!