kiibohd/configurator #111: Unable to flash when removing a default key from layer 1

shinncchen, 2019-01-15:
I was able to flash correctly with kiibohd v0.4.1 but I am unable to flash with kiibohd v1.0.0.

When you start with a default layout for whitefox and a remove a key from layout 1, for example removing 'DEL' layout 1, it tells me 'Compilation failed.' I am unable to see the log.

You can replace all default layout 1 keys but you cannot remove them. For example I would like to remove the default layout 1 key 'DEL', 'VOL-' and 'VOL+' because they are kind of irrelevant but it seems to cause a crash on the compilation.

einalex, 2019-01-18:
I have the same experience with the Ergodox upon changing keys. log:

linux-gnu/Linux is supported.
Unix Build
Selected Generator: Ninja
/tmp/tmp.8VMF1ZPAZd/left
VID: 0x1C11 (Boot: 0x1C11)
PID: 0xB04D (Boot: 0xB007)
-- Build CPU Detected:
x86_64
-- Build Kernel Detected:
Linux 4.15.13-x86_64-linode106
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk2x
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/kinetis.c;Lib/delay.c;Lib/entropy.c;Lib/gpio.c;Lib/periodic.c;Lib/storage.c;Lib/time.c;Lib/arm_cortex.c
-- Bootloader Type:
dfu
-- The C compiler identification is GNU 6.3.1
-- Check for working C compiler: /usr/lib/ccache/arm-none-eabi-gcc
-- Check for working C compiler: /usr/lib/ccache/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Full Compiler:
/usr/lib/ccache/arm-none-eabi-gcc 6.3.1
-- Detected Scan Module Source Files:
Scan/Devices/ISSILed/i2c.c;Scan/Devices/ISSILed/led_scan.c;Scan/Devices/MatrixARMPeriodic/matrix_scan.c;Scan/Devices/STLcd/lcd_scan.c;Scan/Devices/UARTConnect/connect_scan.c;Scan/Infinity_Ergodox/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/kll.c;Macro/PartialMap/layer.c;Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c;Macro/PixelMap/pixel.c;/tmp/tmp.8VMF1ZPAZd/left/generatedPixelmap.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/Interface/output_gen.c;Output/USB/output_com.c;Output/USB/output_usb.c;Output/USB/arm/usb_desc.c;Output/USB/arm/usb_dev.c;Output/USB/arm/usb_joystick.c;Output/USB/arm/usb_keyboard.c;Output/USB/arm/usb_mem.c;Output/USB/arm/usb_mouse.c;Output/USB/arm/usb_rawio.c;Output/USB/arm/usb_serial.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/latency/latency.c;Debug/led/led.c;Debug/print/print.c
-- Found Git: /usr/bin/git (found version "2.17.1")
-- Found LSB_RELEASE: /usr/bin/lsb_release
-- Build OS Detected:
Ubuntu 18.04.1 LTS bionic
-- Found Ctags: /usr/bin/ctags-exuberant (found version "5.9~svn20110310")
-- Found PythonInterp: python3 (found suitable version "3.6.7", minimum required is "3")
-- Checking for kll
-- GITHUB_APIKEY was set.
-- kll Version Detected: kll 0.5.7.3
-- kll Installation Path: /root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.6/site-packages/kll
-- kll Layouts Cache Path: /tmp/7-hid-io-layouts-c5410a1
FATAL Could not find 'MDErgo1-Default-0.kll' DefaultMap
FATAL Could not find 'MDErgo1-Default-1.kll' PartialMap
FATAL Could not find 'MDErgo1-Default-2.kll' PartialMap

einalex, 2019-01-18:
Further testing seems to indicate that clicking the 'clear' button at any time (regardless of whether it clears a key or not) breaks JSON generation as well as compilation.

einalex, 2019-01-18:
further problem: when avoiding the 'clear' button, the program still flashes the default layout instead of the one configured.

xnaas, 2019-02-09:
If you don't want to mess with any JSON, you can simply map `NONE` under `Special`. It's less than ideal, but if you didn't want any function mapped anyway (you're clicking "CLEAR") then this seems like an ok workaround.