kiibohd/configurator #112: Kira (99 key version) LEDs Flicker Periodically

jamiehs, 2019-01-18:
I'm having an issue after flashing the keyboard with the configurator (v1.0.0).
If the LED brightness is anything higher than 0, the row of number keys above the QWERTY row from `1` all the way over to `0` will flash bright white very briefly. Additionally on the first day that I got the board, I plugged it in and left it alone for a few hours; after returning, the number row was colored blue and all of the other keys were not illuminated; at this point, all LED functions of the board were unresponsive.

Today I applied the latest firmware updater (https://github.com/kiibohd/kiidrv/releases/download/v1.5.3-kiidrv/kira-firmware-v0.5.3-installer.exe) and then re-flashed my key map using v1.0.0 of the configurator and the issue still persisted.

I have observed this on a Windows 10 PC using the supplied USB A to USB C cable (plugged directly into the PC) and also on a Macbook Pro (Touchbar) with a USB C to USB C cable. It almost looks like the keyboard power cycles very quickly, but the actual function of the board is unaffected.

Here is a video that I took (slow motion, 1/4 speed) that shows the issue. Note that the general LED flicker that can be seen at the edge of the underlighting is normal when filming in slow-mo, but the bright white flash is the actual issue I'm seeing. The interval of the flash seems random. I had to film the board for about 2 minutes in slow-mo mode to capture it on camera, but I have seen it happen 3-4 times in rapid succession while working today.

https://streamable.com/6e8jg

Is this a known issue? Or might I have a defective board that needs to be returned?
Possibly related, I'm getting a high pitch whine from the board when the LEDs are on. Its volume is proportional to the brightness of the LEDs.

jamiehs, 2019-01-18:
@haata Thanks for the response!

> does the same thing still happen when you have a rainbow animation?

I have not spent enough time with the board to be sure, but if I had to say right now... it seems to only have the white flash issue with a static pattern (like the built in orange one, or one that I create via the configurator). While I've been typing this, it has been on the first rainbow pattern (animating) and it has not flickered (so that's promising).

The high pitch whine happens with any LED state (animating or not, underlighting only, or underlighting + backlighting). The whine does not seem to get picked up by my camera's mic, but if I put my ear to the board I can hear it louder, and when I toggle the LEDs with `RCtrl` + `RAlt` + `PrintSrc/Delete` the whine starts and stops with the lights turning on and off respectively.

After writing all this, I tried recording it with my linear recorder and it seems it _can_ be heard on [the recording (untouched WAV file)](https://www.dropbox.com/s/bfksulbykp8qqpr/190117_0329.wav?dl=1)! This is me toggling the LEDs on and off, then I choose an animated pattern (you kind of hear me switching hands) and then I toggle it a few more times. For me to hear it in the playback, I had to use headphones.

> do you have a powered USB hub to try out? sometimes flickering issues are caused by not enough current available from a USB port

Yes. That is the first thing I tried actually. I have an Anker hub that is powered and when the issue first happened, I tried plugging it directly into the back of my PC instead of the hub as I thought this would help the whine (grounding maybe?) and potentially the flicker too. Unfortunately this did not help. I've built a bunch of projects using WS2812b modules and strips before and have never heard them whine, so I'm stumped on that one... Also a keyboard isn't AC, or high voltage, and [that's where you'd typically hear a whine (from what I understand)](https://www.enlighted.com/blog/why-i-don-t-offer-el-wire-anymore). It's really weird to hear a whine from something that's 5V DC. Just for fun, [I plugged the board into my USB ammeter and it's pulling 5V and 0.5A](https://imgur.com/Ibyslaw.jpg), so that's normal and even _conservative_ for this many RGB LEDs. The port it was plugged into was USB 3.0, so that should be good for up to 0.9A _minimum_.

> does the same thing still happen when gamma correction is enabled? (Press RCtrl+RAlt+Backslash)

This seems to make no visible change for me. I don't see it in the keymap from the 1.0.0 configurator and nothing seems to happen with the LEDs when I enter that command.

> do you have another computer to try with?

I just tried it on my 2014 MacBook Pro (the one that still has USB A ports and a SD card slot) and it also does the white flicker shown in the above video.

> Which OS are you using? Laptop?

* Windows 10 PC (custom built, i7 a few years old, but solid) Tried in both onboard USB (motherboard USB connectors) and on a powered hub connected to it.
* MacBook Pro (2017/2018 touchbar/USB Type-C model) High Sierra
* MacBook Pro (Mid 2014 model) Mojave

> higher brightness, higher whine? or lower brightness, higher whine?

Higher brightness, higher whine

> Also which Kira configuration do you have?

Plastic, black, box royal switches.

Thanks again for helping me dig into this. I'll leave the board on an animation and monitor it to see if I find any strange behavior.

jamiehs, 2019-01-18:
A quick update...
I left the board animating a rainbow pattern last night and when I came back to it this morning it was fine. I used it for a couple of hours and the LEDs did nothing strange.

I eventually used `RCtrl` + `RAlt` + `F1/F2` to choose solid colors and an hour or so after that the colored number row glitch happened again!

[image]

It was red this time, and after it appeared as in the photo above, there was a slow, staggered sweep of two LEDs turning off, so it was like a dark spot went from left to right across the keys, then all the RGB controls were locked up. I had to unplug/replug the board to get control over the LEDs again.

So it looks like the issue has something to do with static patterns in the LEDs, and does not seem to affect animated patterns.

I think my next effort to nail this down as a static pattern issue would be to make a simple breathing animation using the configurator and see how that goes.

vitaflo, 2019-01-23:
I have the same issue, happens on both Mac and Windows, and strangely, there is no animation running.  It seems intermittent.  Random keys will spontaneously flash quickly.  Unplugging and replugging the keyboard can fix it, but I haven't figured out why it starts or what the method to the madness is.  I just know it happens.

haata, 2019-01-31:
Linking this here as it's a firmware issue. It should be fixed for the upcoming v0.5.4 firmware release coming out this week.
https://github.com/kiibohd/controller/issues/307

I'm going to be tracking the whining issue separately (not 100% sure they are related at this point).
https://github.com/kiibohd/controller/issues/310

machine56, 2019-02-12:
ok, basically i have the same exact issue with jamiehs. constant flickering and high pitch whining sound. 
hopefully there will be an update. 

edit:  the high pitch sound occur when full led is on.  and the flickering is happening after i upload my first custom led layout to the keyboard via configurator. 

please help.

miles-prower, 2019-03-21:
The LEDs flickering after uploading your first static visualization is due to this bug described here:

https://github.com/kiibohd/configurator/issues/122
https://github.com/kiibohd/controller/issues/337

The easiest workaround is to export the JSON, delete the completely useless comment, and re-import the JSON — before flashing the keyboard.

haata, 2019-04-05:
Any more issues related to this?
I believe most of the firmware issues have been worked out. Though there is still a chance that old animations (created before the most recent firmware update) can interfere with the new animations and cause flickering. In this case you'll either need to fix those animations, or perhaps easier would be to remake the animations.

yokhoe, 2021-02-01:
Hi, was wondering if this issue is still ongoing? I don't really change my LED setup other than the default rainbow animation.  Occasionally it would freak out (I noticed after a long session) and either [the LEDs] turn off completely, or only select keys are lit up... I'm not sure if this issue is related or if anyone can point me to the right direction. 
[image]

haata, 2021-02-01:
There can be a few issues:
1. Bad USB cable
2. EMI interference from a nearby device (like a cellphone/microwave/short wave radio/etc.), this helps a lot: https://kiibohd.github.io/wiki/#/KiraRF