kiibohd/configurator #115: Kira Lock LED's do not illuminate

vitaflo, 2019-01-23:
I have a strange issue where the lock key (caps, scroll, num) LED's do not illuminate on my Mac.  When plugged into a Windows 10 machine they do illuminate, but intermittently (after a computer restart they may not function, but unplugging and replugging in the keyboard will fix it).

I have no LED animations running, and the LED's are on (not set to off with the special functions or whatever), however they also never turn on if I do have animations running either.  Strangely, when I do an LED test when on my Mac, the lock keys DO illuminate.  They just never do during normal operation.

Epod, 2019-01-24:
I also have this issue, however the LED lights never successfully illuminated for me. Default settings out of the box with updated firmware.

sawman, 2019-02-02:
For me the lock LEDs don't illuminate only if the global LED brightness is set to 0. The brightness of the lock LED seems to sync with the backlighting brightness setting

haata, 2019-02-03:
This will be fixed in the upcoming v0.5.4 firmware.

https://github.com/kiibohd/controller/issues/313

rondog, 2019-02-05:
@hooke007 I just flashed with 0.5.4 as well and can confirm mine are not lighting up either with my mac

haata, 2019-02-05:
I just test on my mac, only Caps Lock LED works. I don't believe I've ever seen Num Lock or Scoll Lock LEDs work on a mac (if you have please let me know and I can investigate further).
In the next release or 2 I'll have ways to show layer indication and other things using those LEDs (they are RGB).

rondog, 2019-02-05:
Could changing the LED behind the key be an option? Not sure if that is any easier or worth exploring.

gokieks, 2019-02-07:
Num Lock LED not illuminating on mac OS applies to every keyboard I've ever used that has an LED for it.  On my hackintosh it will come on during BIOS boot, and may stay on when mac OS finishes booting, but if you toggle Num Lock again after that it will remain off.

MylesShannon, 2019-02-07:
> This will be fixed in the upcoming v0.5.4 firmware.
> 
> [kiibohd/controller#313](https://github.com/kiibohd/controller/issues/313)

Caps lock LED is now working for me on MacOS after a firmware update today, presumably v0.5.4

haata, 2019-04-05:
The update today (v0.5.6) should help with this.
There are still some reports of non-working LEDs but I haven't been able to reproduce them yet.

macOS NumLock and ScrollLock LEDs are not used so will not light up.