kiibohd/configurator #119: Multiple Kira Issues

ChitownM2, 2019-01-30:
Received my Kira on Saturday and finally had a chance to take it out of the box tonight. First thing I did was go to the kono store and follow the instructions to download the drivers & update the keyboard using the configurator as instructed by the manual. I started with the standard layout and then downloaded the firmware to the board by entering flash mode by pressing Shift-Ctrl_ESC. Everything updated smoothly.

I then started looking at the lighting control options. Nothing seemed to be making sense based on the layout I had flashed and the actual responses I was seeing on the board so I tried to update the firmware again but instead I used the firmware tool linked in the manual. After installing the latest firmware using that tool, I now have the problem that it explicitly states it fixed in the firmware update. The first couple times I updated the layout from the configurator I used the CTRL-SHIFT-ESC keys to enter the flash mode and everything worked fine. After using the firmware update tool, using those keys to activate flash mode causes the shift key to be permanently pressed on the system even after the keyboard is unplugged and requires a system restart to fix. 

Also, with the latest firmware, the LEDS no longer turn off and have the 1 light up orange on the bottom to indicate it is in flash mode. Instead the LEDS basically just "freeze" when it goes into flash mode. Thus if you have the backlight set to a solid color rather than wave it is impossible to tell if you are in flash mode or not. This only happens if you use the reset button to enter flash mode. Using the key combination does produce the LED effect described for flash mode, however it causes the "shift" problem so that means it's not really an option.

For lighting control, there are several discrepancies between the manual and the functionality that I am seeing. According to the picture on the manual, key fading should be controlled by F3 & F4 on layer 2 for backlight and underglow respectively and this does work on the board. However, on the "standard" layout I flashed to my board it shows both of those keys as unassigned in the configurator. I also can't find how to assign those functions to any other keys using the configurator as fade controls don't seem to be included anywhere. Same goes with adjusting the backlight or underglow color using F1 & F2 respectively. They don't show up in my keymap and can't be re-assigned....

The manual & the default layout I flashed both say that I should be able to save an animation and brightness configuration using F12 on layer 1. It also says that I should be able to bring up the default board animation setting using F10 on layer 1, but neither of these commands does anything. I can't save a configuration, can't load the default and can't load the saved config after I change the animation mode using F7 on layer 2.

In addition, the manual says that the settings should be retained when the keyboard is unplugged, but this is not the case. When I unplug the board and plug it back in it defaults back to the "miami wave". I'm guessing this is tied to the issue above about not being able to save???

Also, is there any way to change how the different layers are accessed? Since a FN key was not included and I don' t want to give up any base layer keys to assign layer functionality, the only way to access layer 1 is right shift + right control and layer 2 is right alt + right control. Can this be customized? Or how can I assign a new combo to access layer 3? For example right shift + right alt would work, but I don't see how to set this in the configurator.

efjacobson, 2019-01-30:
experienced many similar issues myself, personally im going to ask around in discord to make sure i understand exactly what is a bug and what is poor documentation, then post specific bug issues here. fwiw, i _think_ you can re-assign the layer functionality via the raw json. the problem is that macros do not show in the configurator once set. however, i have seen them back when i was flashing back to default regularly.

if you arent already too invested in your layout, give that a try - reset to default by getting the `layouts > standard` firmware, and see if you see/edit the macros then. you can also see them in the json, look for `Macro 'Layer 1'`

ChitownM2, 2019-01-30:
Thanks for the input. I think the shift key issue is definitely a bug so at least some of my issues are genuine. 

I was waiting to setup my layout until I had everything working so I did go back to the configurator and load up the standard layout again and do see the macros now. That definitely answers that particular question so I updated my original post, but the fact that they don't show up in the configurator after the initial loading might also be a bug. 

The json does seem to indicate that F10, F11, & F12 should function to default, load and save animations so I'm not sure what is going on there. 

I don't see any mention in the json for the layer 2 functions of F1, F2, F3 & F4 though. I only see the grid reference when it defines the keys and then again when it scans the grid.

ChitownM2, 2019-02-05:
> I am also experiencing issues. When I try to use the load/save functions to store a specific lighting combination, it sometimes works and sometimes doesn't.
> When it doesn't work, the LED will somethimes flash frantic when I try to load.
> When programming a new firmware with a new static LED configuration and setting this as the "Starup Animation", I also get the frantic flashing LEDs.
> If I then cycle through the available animations, I never run into the one that I programmed.

So my original post here is in the wrong repository. Nearly all of the issues I've described are an issue with the Kira firmware and not with the configurator. Most of the issues are already being addressed and can be viewed [here](https://github.com/kiibohd/controller/issues).