kiibohd/configurator #120: Configurator Stuck Loading on Kira

GnarleyCosmonaut, 2019-02-05:
On windows 10 I am unable to access the kira page in the configuration tool. It is stuck "loading" even after refreshing, reopening, restarting the computer and trying again. I gave the tool over an hour to load and it still did not come up.

GnarleyCosmonaut, 2019-02-11:
Getting a connection_timed_out error when trying to pull the Kira-Standard.jason file. Does not seem to be a firewall issue.

tomplex, 2019-02-13:
I'm getting the same issue on MacOS 10.13.6.

simplexion, 2019-02-15:
I'm seeing this issue in Windows 10 1709 as well. This is while running Kiibohd as Administrator too. IN Linux (running as root) it works fine.

bjohnso5, 2019-05-03:
I am also having this issue, but in my case I think it IS a corporate firewall issue. Is it possible to host the layout files on standard http port 80 or even better, over https on 443?

travVT, 2019-06-04:
I'm having the exact same issue with my Kira keyboard on Windows 10. @bjohnso5, did you ever have any luck solving this? Thanks!

bjohnso5, 2019-06-04:
@travVT no, sorry, I didn't want to spend too much work time looking into it, as that's where I use my Kira.

Kyle-Falconer, 2019-07-25:
Similar issue on macOS Mojave, with the K-Type keyboard.

I'm getting a `net::ERR_CONNECTION_TIMED_OUT` for "versions" and "KType-Standard.json"

[image]

Kyle-Falconer, 2019-08-06:
On further investigation, I have confirmed that I'm unable to access that URL that is requested by the software:

```
http://vash.input.club:3001/versions
```

On my work laptop, I can access `vash.input.club` but not over port 3000.

On my home computer, I can get a response from that URL over 3000 just fine.

tomplex, 2019-08-12:
For me it was my corporate network blocking certain ports. I ended up connecting my computer to my phone's wifi hotspot and re-trying; it worked just fine. You should either try to configure your keyboard at home or using some other network, if you have access.

Kyle-Falconer, 2019-08-12:
The fix for this should be to re-deploy the API for that endpoint to a more standard port 80 or port 443. That way, it would have a much lower chance of getting blocked by corporate policies.

Kyle-Falconer, 2019-08-12:
@jbondeson do you know how this endpoint might be able to be moved to a standard port? It seems you were the last person to set this:

https://github.com/kiibohd/configurator/blob/1ba70df9046443ac465e1c2e801025ad1f68b4cc/src/renderer/state/settings.js#L13

Kyle-Falconer, 2019-09-24:
It's working for me now with 1.0.2 and commit 18ac1cd950fb16c0985da50340d2560a0b525022

Thanks @jbondeson !!! This might have been a "small issue" for you, but it was a total blocker for those of us affected.

travVT, 2019-09-24:
1.0.2 is working for me too! Thanks @jbondeson. :)