kiibohd/configurator #122: Configurator screwing up static LED Visualization JSON

nathan-eko, 2019-02-11:
I go setup a static visualization. I select it as the startup animation. I flash my Kira. It does exactly what I think it should have.

I go change some other things around. Doesn't really matter what they are. The next time I flash the Kira, it's doing a blinding strobelight thing that never stops. The only way to get it back to a working state is to delete my static visualization and start all over again. 

What's different between the working state and the epilepsy inducing state:

[image]

I can fix this JSON manually and "import" it, but the Electron app continues to jack it up the same way over and over again.

How about we don't put comments in our JSON at all?

fspinillo, 2019-02-12:
I'm getting the flashing strobe light after modifying anything related to the lights. If I just modify the keys, the keyboard doesn't run into an issue. As soon as I modifying anything with the lights it goes nuts until I hit the rest key again.

nathan-eko, 2019-02-13:
You have to go delete that useless embedded comment every single time. The firmware seems to interpret any line it doesn't understand as a 255, 255, 255 on all LEDs and even when the type of display is `static` it sits there cycling through the lines over and over and over. If someone honestly thinks this warning is necessary (like it's somehow *less* dangerous to edit a non-static map?) then they should conditionally display it in the UI based on the presence of the `static` type in the JSON instead of trying to trick it into showing up somewhere it doesn't belong.

That said, this app spews out invalid JSON like crazy. If I delete the default layer 1 mapping on the escape key it can no longer "compile" at all, the "show log" link does nothing and I can't even view the JSON to see what's wrong with it. There are dozens of Node modules embedded in it but not one of them is a linter. 👍

Bat-Shark, 2020-02-28:
Still hitting this flashing a Kira last night. That comment caused the flash to succeed but the startup animation was basically dropped as a result instead of strobing lights.

akbyrd, 2020-03-04:
The configurator _really_ needs to produce an error message when compilation fails.

huguesalary, 2020-04-08:
I'm also running into issue where the compilation silently fails... very frustrating. Until an actual log  is available in the interface, can someone provide a way to manually compile and get the debug output?

Bat-Shark, 2020-04-08:
I got around it by removing the comments in that section. Doesn't help with
an actual log but gets you working.