kiibohd/configurator #126: Fingerprints not working on certain keys on Kira

mattkemp, 2019-02-20:
I'm using the fingerprints_two_tone animation on a Kira, but it's not affecting certain keys (down arrow, right arrow, num0, and num.)

The LEDs are definitely working. Also if I change to a wave animation they do change.

Is there something special about these keys?

gowness, 2019-03-12:
It looks like this is due to these lines of custom kll that are generated: 

```
#~~~START ANIMATION (fingerprints_two_tone) INJECTION~~~#
S[0x00-0x5F] :+ A[fingerprints_two_tone](start);
#~~~END ANIMATION (fingerprints_two_tone) INJECTION~~~#
```

but 0x5F is the left arrow on kira, and the rest of the keys go to 0x63

```
#~~~START ANIMATION (fingerprints_two_tone) INJECTION~~~#
S[0x00-0x63] :+ A[fingerprints_two_tone](start);
#~~~END ANIMATION (fingerprints_two_tone) INJECTION~~~#
```
this fixed it on mine, though i suspect it will get undone if i update the animation

haata, 2019-03-14:
This should be fixed now https://github.com/smasher816/kiisrv/commit/432908a890e16771dbe0404224ac994748a35344

closed 2019-03-14 by haata