kiibohd/configurator #128: Custom Layout upload failure

msz006c1, 2019-02-27:
Upload custom layout for Infinity Ergodox. After flashing the new firmware, layout appears to be bundled default. Double checked downloaded firmware JSON, confirms the custom layout was not uploaded for compilation. Instead, original default layout was compiled and downloaded for flash.

ibaranov-cp, 2019-03-03:
+1 same issue

ilvadol, 2019-04-15:
The same, but on Kira.

I've found out that if you clear any of the preassigned commands in layer 1  or 2, it gives the Compilation Failed error.
If you add command to any unassigned key on those two layers (1 and 2), it spits out the same error.

Here is the default layout, which compiles no problem (I can change base layer without issues):
[image]

And here I cleared "M" key on layer 1 (got rid of "Menu"):
[image]

Here I reset to default layout and added the "Play" function to "N" key on layer 1:
[image]

Additionally, I can't view logs when this error occurs (the Internet connection is stable. 

The issue persists on both Windows and MacOS. This is really sad, because I'd really like to customise the default layout (layers especially).

mchaput, 2019-05-05:
Using a Whitefox on Mac, I find the following:

- If I try to make more than one change at a time, it always fails to compile. (For example, if I load a working map, then make more than one change and try to compile, it will fail.)

- If I try to clear an assignment on Layer 1, it fails to compile. I can *reassign* a key on layer 1, but not clear it.

- Whenever it fails to compile, the "Show log" button on the "Compilation failed" banner does nothing.

rbuchberger, 2019-05-28:
I'm having the same issue as well. White fox keyboard on arch linux, tried both the appimage and the AUR client. Sometimes compliation works, sometimes it doesn't.

MJohnson159, 2019-06-10:
@mchaput I'm trying to flash my old infinity 60 and had to download this and your comment helped me understand wtf was going wrong. It takes like 10 seconds to compile a change and I can only change 1 button at a time (currently trying on layer 1) to get it to compile. I'm trying to change my arrow keys from WASD to ESDF, which it already is on my board, but I'm trying to change something else and get the layout to my current layout and then update my desired changes.

If I remove UP from W and LEFT from A, it freaks out. Meaning the same issue mchaput said above: removing an assignment on a layer breaks compilation but reassigning it doesn't. 

How is this still a bug? If this happens to other keyboards, how do people even make their own function layers if this prevents any compilation for anything other than a default layout? PLEASE get this working as I don't have the tic mark from my original flashing and I can't flash my current layout back onto my board with this in order to add the tic mark under my ESC key.

MJohnson159, 2019-06-10:
@smasher816