kiibohd/configurator #129: Custom Animation Preset...

rynodigital, 2019-03-01:
Is it possible to have custom animation where all the LED's on the keyboard off _unless_ a key is pressed, and then the Key fades off again?

mattkemp, 2019-03-01:
Have you tried a fingerprints_two_tone with the second colour being black (#000000 - all LEDs off) ?
Combined with the current bug where static colour startup animations don't trigger when you unplug and plug the keyboard back in, you wouldn't even need to set the startup to a custom colour of black.

rynodigital, 2019-03-04:
This worked!

I have a Fingerprint two-tone like you suggested with the second color being black

For the ’Startup Animation’ i’ve got a ’Static LED’ animation set to black, so that everything is off by default.

Thanks for your help on this, loving the board!

RYAN SANDLER <>
480 205 1102