kiibohd/configurator #134: Kira defaults to NKRO on boot (does not work in BIOS or GRUB2)

jiggahertz, 2019-04-08:
My Kira (with latest firmware) does not work to interrupt boot or use GRUB2. The issue is that my Kira defaults to NKRO when my computer powers on, which causes issues with BIOS and bootloaders. To avoid these issues, the Kira keyboard should default to 6KRO when it boots and only switch to NKRO only after entering the operating system. Fixed with [this suggestion](https://github.com/kiibohd/KiiConf/issues/70#issuecomment-271423622) but still requires manually switching back to NKRO if user wants to use it once booted into the OS.

haata, 2019-04-08:
Interesting.
I'll need to dive into the Grub2 source base at some point. I'm almost certain that grub has a bug where it does not properly do the negotiation of 6kro vs. nkro that's required in the USB HID spec.

(As an aside, Linux also has this bug and I've been poking around the kernel a bit to see how a patch can be written for the kernel, likely right here in this function: https://elixir.bootlin.com/linux/latest/source/drivers/hid/usbhid/usbkbd.c#L273)

haata, 2019-04-08:
For future reference, this is probably the file in grub2 where the keyboard should be negotiated back to 6kro http://git.savannah.gnu.org/cgit/grub.git/tree/grub-core/term/usb_keyboard.c.

ciphermenial, 2019-07-12:
Will I always require a second keyboard to do anything in UEFI and grub because it feels like a lot of these problems are never going to be fixed?

haata, 2019-07-13:
So, you can switch to 6KRO mode, that will work in the bootloader.

I have also looked at the grub code and found where the bug is. But that has little to do with the keyboard and more to do with UEFI and Grub not following the USB spec.

ciphermenial, 2019-07-16:
That is a workaround and an excuse. The issue I see here is I haven't come across another keyboard that fails to interact with firmware or grub.

Which of these sounds more reasonable:
- Grub maintainers resolving an issue affecting a tiny amount of users
- Hardware manufacturer resolving an issue that affects all users of their product

I believe @jiggahertz suggestion above sounds reasonable.

haata, 2019-07-17:
The problem is that the USB HID spec does not define any such way to know an OS is booted.
All that the device knows is whether or not the OS has requested a boot device.

Grub, Linux, Windows and UEFI do not follow the spec (macOS does), and just assume every keyboard is 6KRO. https://www.usb.org/sites/default/files/documents/hid1_11.pdf Section 7.2.6 

The reason I have been working on HID-IO is to make it possible to query this sort of information about the computer. This interface was designed way back in the early 90s and a lot of bad decisions were made.

All other keyboards you're trying out likely only do 6KRO at best. This issue affects QMK keyboards as well.

Anyways, until HID-IO is ready, I'd recommend setting a configuration option from the configurator which will default the keyboard to 6KRO. 

[image]

If you need to switch back to NKRO, I'd recommend mapping a key to: 

[image]

ciphermenial, 2019-07-18:
Thanks! That is much more useful information. This workaround seems pretty simple to implement and is worth it considering how irregularly I access firmware or grub.

razamatan, 2020-04-25:
yeha.. i ended up just using the toggle and keeping the default state nkro.  the off chances i need to interact w/ bios/bootloaders, i toggle over to 6kro.  the reinit of the usb on os boot has my infinity ergodox back at nkro when it's work time.