kiibohd/configurator #135: Custom Ergodox LCD

slyrat, 2019-05-27:
This is more a feature request to add the ability to have custom images for the left / right lcds for the Ergodox. Something as simple as converting a bmp to the appropriate text for setting in the defines could work.

LordHiler, 2019-07-18:
I would also appreciate this, as well as the ability to set the backlight color per layer (including the base layer).

typorian, 2020-03-29:
I have some software called bitmap2struct.py, which seems to be copyrighted by @haata himself. I have no idea where I got it from though, but it can be used to make appropriate bitmap files into LCD definitions that will then be displayed on the respective layer. You can do something like this with it: http://imgur.com/89EDUQQ