kiibohd/configurator #137: Ergodox Infinity: Compiled firmware doesn't work properly

thegreatco, 2019-06-21:
When I attempt to compile firmware for the infinity (both on the webiste and using the Kiibohd Configurator app) the resulting firmware ends up in a very weird state. Its like the keyboard is stuck in some combination of layer 1 and the base layer. For example, my base layer is a fairly standard keyboard layout, with the lower vertical button on the left set to bring up layer 1 (not a lock, just a shift) with the key in the upper left of that side being set to "Flash" in layer 1. Without pressing any other keys (no layer shift) pressing the upper left button puts the keyboard into flash mode. Additionally, most of the other keys appear to be shifted to the right by one column.

djak250, 2019-09-16:
Same except flashing the default bootloader and keymaps from the link you posted got me up and running. Any custom keymaps though and it's back to SOL.

haata, 2020-03-24:
The LCD screens have been fixed with the most recent firmware update.

closed 2020-03-24 by haata