kiibohd/configurator #140: Infinity Ergodox Screens

LordHiler, 2019-07-18:
Latest configurator firmware does not correctly display layer changes on screens for Infinity Ergodox. This is a major issue for me and I'm sure many others. Please fix.

sgtmjrme, 2019-08-09:
Agreed - I've had to go back to a much older build to get this to work properly, but then my LED's on the board aren't working.

woodlander87, 2019-08-20:
I am also experiencing this. At first I thought that the layers had not been applied at all.

Phiphel, 2019-09-07:
I have the same issue.

djak250, 2019-09-16:
Same here. And then I got into a horrible spot where I can't flash any of my keymap and have to use the default bootloader and kemap. :(

ur5us, 2019-10-16:
@haata Any news on this one or #137?

NFiorentini, 2019-10-27:
Same issue. Except that reverting back to v1.0.1 Release Candidate 1 did not resolve this issue.

jeffutter, 2019-10-30:
This is pretty bad. I hadn't updated my Infinity Ergodox for some time. Last time I updated it there was some web GUI to generate the configs, which I can't find anymore. I wish I had found this thread first. This keyboard has been my daily driver for years and now is nearly unusable.

kannony, 2019-11-04:
Yeah, also reporting that the LCD doesn't display the current layer. This is a launch model Ergodox Infinity.

matthewh, 2019-11-08:
Same here. The LED does not update when switching layers

kannony, 2019-11-08:
> Same here. The LED does not update when switching layers

I'm reading that there are LTS versions of the firmware that do not have these issues. However, I haven't the faintest idea on how to find those LTS versions. :-(

laumann, 2020-03-08:
Just chiming in to confirm that this is still an issue. I flashed my infinity ergodox yesterday using [v1.0.2](https://github.com/kiibohd/configurator/releases/tag/v1.0.2) of the configurator, and everything works fine _except_ that the layer changes are no longer showing on the LCD screen(s) (the layer changes are active though - it's just the LCDs not showing anything).

This model is a couple of years old, was from a group by on Massdrop, not sure if that helps.

laumann, 2020-03-08:
> Agreed - I've had to go back to a much older build to get this to work properly, but then my LED's on the board aren't working.

@sgtmjrme Which version did you go back to?

kannony, 2020-03-08:
@laumann I [found a firmware for the Infinity that](https://github.com/cweagans/ergodox-firmware) @cweagans was kind enough to upload.

I see instructions for flashing the firmware but haven't been able to fully make sense of them since they require putting the keyboard into flash mode and THEN typing commands into a command prompt.

But more or less, the instructions are simple:

1. download the cweagans firmware.
2. Use DFU-UTIL.EXE to flash the right side of the Infinity.
3. Use DFU-UTIL.EXE to flash the left side of the Infinity.

The problem is that I haven't the faintest idea how to put just the right-side into firmware update mode. So that's where I dead end.

There is also the official [firmware repo located here](https://github.com/kiibohd/controller/releases).

cweagans, 2020-03-08:
> they require putting the keyboard into flash mode and THEN typing commands into a command prompt

Correct. It's annoying and requires a separate keyboard. I'm using a macbook, so I happen to have an extra keyboard built in.

> The problem is that I haven't the faintest idea how to put just the right-side into firmware update mode. So that's where I dead end.

Plug the keyboard half in directly to your computer, flip it over, and push the button labeled S39 on the PCB. It can be found here:

[image]

If you're using the acrylic case that came with the Infinity, there's a small hole drilled directly over the switch, so you _should not_ have to take your keyboard apart. Once you push the button, the LCD backlight will turn red and that half of the keyboard will be in flash mode and you can proceed with the instructions.

Hope that helps!

kannony, 2020-03-09:
@cweagans hey thank you so much for that! I had completely forgotten about those reset buttons! (I'm extra stupid because I had to drill them myself). But I can confirm that your firmware has functioning LCDs for moving through layers.

The only issue that I'm having is applying my custom keymap to your firmware. For whatever reason, I can't get the keymap to work. I've tried importing it but nothing seems to stick.

cweagans, 2020-03-09:
🤷‍♂  can't help you there. I just used the configurator way back when. I haven't updated my keyboard firmware since I published that repo, so that's still my daily driver.

haata, 2020-03-14:
Please try this version of the beta firmware and see if this helps (I'll push a release to the configurator once I get some confirmation that this helps).
https://github.com/kiibohd/controller/releases/tag/v0.5.7-beta1

AlexBaldwin1, 2020-03-15:
@haata I flashed your linked beta firmware and it fixes the LCD screens and the layer locking issue I was having.

kannony, 2020-03-17:
@haata thank you! That firmware works! The LCD screen colors change when switching layers and it displays the correct layer. 

I am sorry to ask this question, but am new to Kiibohd: importing a keymap doesn't seem to work. When I try to import my keymap, the keymap does not stick; the keymap resets to default after flashing the beta firmware. I also tried setting my own keymap and then flashing but the keymap didn't stick then either.

In case the keymap installation procedure matters, here's what was done:

1. copied my keymap to JSON in Kiibohd
2. imported keymap in Kiibohd (I don't know if this makes any sense to do)
3. flashed right half using Kiibohd and unplugged
4. flashed left half using Kiibohd
5. power cycled both sides

Afterward, the keymap is the default.

haata, 2020-03-17:
@kannony The firmware isn't live on the configurator yet, I'll make sure to update everyone here when it is. The beta firmware only has the default layout.

I found an annoying bug in the new firmware that I'm working through and it's holding me back on the release. It's not Ergodox related (fortunately), but it's tricky to only update firmware for a single keyboard at this point. Hopefully I can resolve it in the next day or so.

Omnicraft, 2020-03-23:
Is a new release of the configurator needed or will the existing configuration pull updated firmware once it's released?

haata, 2020-03-23:
I'm working on pushing the new release to the configurator (having some technical issues with the server right now). I'll update this (and various other bugs) once the release is pushed.

haata, 2020-03-24:
Firmware update has been pushed. I'll close this issue, but please feel free to provide feedback here.

closed 2020-03-24 by haata

laumann, 2020-03-25:
@kannony Thanks for the info! And thanks @haata for fixing this: This means I can just use the latest configurator?

Omnicraft, 2020-03-25:
Yeah, the configurator pull the firmware at compile. Since it the push I was able to flash my ergodox infinity with a custom layout and working LCD labels and color

laumann, 2020-03-25:
Ah, so I need to build the configurator from source, there's no new release yet?

Omnicraft, 2020-03-25:
nope, just click compile like usual. if that doesn't work then I'd suggest
reinstalling the latest configurator. Worked for me.

kannony, 2020-03-25:
@Omnicraft I've tried reinstalling the configurator then importing the JSON for the keymap but when I try to flash it, I get the error: "Error Parsing: Unexpected token in 'in JSON position 0"

It sounds as if I will need to manually set a keymap then?

(Thanks by the way for fixing the LCD and layer issues, by the way, it's great finally being able to see what layer I'm on!)

Omnicraft, 2020-03-25:
@kannony your saved json may no longer work with the updates to the KLL and configurator, depending how old it is. I had to make my layout in the configurator again.

laumann, 2020-03-26:
Oh, I didn't realize the compile button pulled stuff automatically, I ended up building a new version of the configurator :smile: Short story: It worked! :tada: (It's actually better than before flashing it the first time, back then only the LED on the side connected to the computer would light up, now they're both lighting up).