kiibohd/configurator #143: K-Type: Successful flash with 1.0.1, then it only works after boot

BinaryHammer, 2019-08-08:
I successfully flashed the K-Type with the 1.0.1 configurator. Then the K-Type will go lights on without registering input, then it will go lights off after some seconds and on and off and so on. But pressing keys will not do anything.

In the configurator you can see that the green dot is there for 3 seconds and then vanishes, every time I go to the home screen of the configurator.

After 3-5 minutes sometimes the keyboards just starts to work and behaves normally.
It only creates problems when unplugging and plugging back in while the OS is running.

Using macOS 10.14.6 on a MacBook Pro 2017 with USB-C to USB-C connection.

tastheimer, 2019-08-13:
I actually have been having the same problem on my k-type on a Mac, but it happened even before 1.00/1.0.1. I suspect that you are experiencing the same issue as me but circumstantially it happened after flashing. I can usually get my keyboard to function again if I plug and unplug it a few times.

BinaryHammer, 2019-08-28:
I do not know why, maybe it is the macOS Update to 10.14.6 or that I deleted the K-Type Pixelmap in Network preferences, but the error is gone. Even though I did not flash the keyboard again.

closed 2019-08-28 by BinaryHammer