kiibohd/configurator #147: configurator 1.0.1 won't flash even with known good JSON

von-zinzer, 2019-09-07:
Hello
Attempting to flash my K-Type with a known good JSON that was loaded from a previous flash. Nothing happens when FLASH KEYBOARD button is clicked. K-Type is in flash mode using Fn+Esc.

haata, 2019-09-07:
There have been some bug fixes in firmware and configurator that may cause old configurations not to work. You may need to remake your configuration.

notJerl, 2019-10-15:
This has been happening to me continuously as well.  This is not because of old configurations.  Earlier today, I started a brand new configuration for my Kira fresh from scratch using the default, unedited Standard layout pulled fresh from the server.  I made my changes, including quite a few remaps, a couple macros, and a static LED visualization. Clicked Flash Keyboard.  The configuration compiled fine and successfully went onto my Kira, so I saved the JSON in a text file for future use and *closed Kiibohd* (this is pretty important).

Later, I decided to make some changes to the keymappings.  I opened Kiibohd, changed the mappings, clicked "Flash Keyboard", and.... Nothing.  No error message, nothing in the logs, it just stopped.  I tried clicking Layouts and clicking the most recently successful build, then clicking "Flash Keyboard".  Nothing.  I tried loading a brand new fresh from scratch configuration using the default Standard layout pulled straight from the server, clicking "Import Keymap" and pasting in the JSON that worked earlier.  Nothing.  I tried clicking "Home", which I've found reverts all changes you've made but haven't flashed and going back in and flashing the supposedly *exact same* configuration that was there earlier.  Nothing.

So, to test my hypothesis, I rebuilt the entire configuration from scratch.  I had most of the important stuff screenshotted, so this wasn't *too* difficult, just tedious.  Clicked "Flash Keyboard", and it successfully compiled the configuration and flashed it to the board.  Then I *closed Kiibohd* again, reopened it, and clicked "Flash Keyboard" with the configuration still exactly the same as it was before I closed the program.  Guess what happened?  *Nothing.* 

My guess is that there's either something wrong with the way it parses the JSON when the program starts up or when you have it import a keymap, or there's something wrong with how it generates the JSON, since that appears to be how it saves the configuration to disk.  There's other evidence for this.  For example, the Kira's base layer by default has two macros set to enable Layer 1 and Layer 2, and they both show up in the macros list.  They're *gone* from the list, even though they still show up in the Custom KLL (where they didn't show up before).  Something is causing them to not be parsed correctly.  This doesn't always happen, and I haven't narrowed down what it is that makes it happen, but it essentially means that for my configuration, I have to either leave Kiibohd running always, or start fresh every time I want to make a change (even if it's just mapping or unmapping a single key).

notJerl, 2019-11-05:
Updating that I've confirmed that my problem seems to be related to #122.  Removing the `### AUTO GENERATED - DO NOT EDIT - STATIC COLOR MAP ###` comment from the animation in the Add/Edit Animation screen causes the configuration to compile.