kiibohd/configurator #151: Can't open configurator app on macOS

dylanyouk, 2019-11-04:
Downloading the configurator app for macOS and opening gives the following message:

> “Kiibohd Configurator” cannot be opened because the developer cannot be verified.

I am using macOS catalina. Also tried downloading the zip file but the app in there just says it "cannot be opened" with no other explanation. any workaround for this?

haata, 2019-11-04:
We'll look into this but there is a work around.
Do a "Command+Click" then select Open, the dialog will allow you to open the executable.

We're looking into the costs/process of signing an executable on macOS.

dylanyouk, 2019-11-04:
thanks! didn't know about this macOS trick (turns out it's actually "ctrl+click" though)

CaseyB, 2020-03-18:
I still get that it can't open when I ctrl+click and choose open.

bennettp123, 2022-07-08:
I'm getting a new error on macOS Ventura

[image]

This seems to only affect the zip file though (https://github.com/kiibohd/configurator/releases/download/v1.1.0/kiibohd-configurator-1.1.0-mac.zip) &mdash; the dmg file works fine (https://github.com/kiibohd/configurator/releases/download/v1.1.0/kiibohd-configurator-1.1.0-mac.dmg &mdash; you still need to ctrl+click and open, but afterwards it launches and works fine)