kiibohd/configurator #152: Hexgears Gemini: Unable to change action of RGUI/RWIN

kentnek, 2019-11-06:
I just received my Hexgears Gemini Dusk and installed the forked configurator for Hexgears keyboards. Before flashing, the button on the right of my Right Alt had action `MENU`, however after flashing it always becomes `RGUI` (Right WIN), regardless of what action I set to it.

WolfLink, 2019-11-26:
I cannot change the action of the RGUI and I also cannot change the action of the fn1 key next to it.

villains, 2019-12-03:
Having the same problem.

Coming from a 60% keyboard that I had configured optimally for my typing style and workflow, I tried putting the same key mappings and layers on the Gemini Dusk. After wasting some time trying various things, I found it's impossible because some important keys (RGUI/RWIN, FN key next to it) cannot be remapped properly.

It's easy to reproduce. Try to map those keys to some characters (e.g. 'A' and 'B') and then go to a text editor and see what happens when you use the keys.

It is irritating that I can't get the much more expensive Gemini to work the same way as the budget 60% keyboard that it was supposed to replace. So the Gemini is now hidden away in a drawer until this bug is fixed.

Hoping you can fix this ASAP. I can volunteer as a tester if that helps.

jaredbidlow, 2020-01-29:
I have an Infinity, Alphabet 60% keyboard that I can't repurpose certain keys, like the capslock and fn2 in the base layer, but probably not limited to those.

mathewng, 2020-10-26:
I just bought this keyboard too and I have this workaround.
Adding the two lines below into **Custom KLL** section in the hexgears configurator defines the MENU and FN keys.

```
S106 : Layer[1];
S108 : U"App";
```

You can amend the right side of the colon to any key you want to output.

WolfLink, 2022-01-27:
How strange.  I am able to successfully rebind those four keys to letters in this way for testing, but binding them to modifier keys has weird results.

The four keys to the right of spacebar can be remapped as S105, S106, S108, and S109.

My desired layout is RGUI, RALT, RCTRL, FN

Programming those four keys as a, b, c, d works as expected.
Programming them as desired results in RGUI, RGUI, FN, FN (I am programming FN as #:Layer[1])
Programming them as LGUI, LALT, LCTRL, FN results in: LALT, LGUI, LCTRL, FN
Programming them as LALT, LGUI, LCTRL, FN results in: LALT, LALT, LCTRL, FN