kiibohd/configurator #153: Hexgears Gemini: ENTER key is mapped to key 70, but actual key is 71

elafargue, 2019-11-07:
On a new Gemini keyboard when configuring animations ,the configurator maps the Enter key to id 70 which is not correct, it should be 71.

How to reproduce: create a single color led layout, flash the keyboard. The Enter key will be unlit.
Fixing: edit the animation manually, replace the mapping at 70 by 71, and it works fine.

al45tair, 2019-11-11:
FWIW, I can't get the manual edit to work. When I change the P[70] to P[71] in the "frames" array in the JSON, the result is all of the lighting on the keyboard turns off. Perhaps I'm missing something?

elafargue, 2019-11-11:
Yes, there are apparently other bugs on top of this, but as long as you just define a static LED mapping, change "70" to "71" and you don't go back to the key mapping screen, then it should work. But if you get fancy after this and try to build a complex mapping, in my experience it breaks down completely and often the keyboard firmware will remain stuck in bootloader mode, or require a press on the reset button to upload a new FW...