kiibohd/configurator #155: Not able to flash custom animation on Gemini

xiletommy, 2019-11-11:
As the title suggests, I cannot flash a custom animation using the Hexgear configurator for my Gemini Dusk. If follow the quickstart on Kono (the one where you choose a pre-set animation and just change the color) and flash the custom animation, my keyboard just goes dead. The only way to make it functional again is to put it back into flash mode via the reset button on the back (key inputs don't work) and flash one of the original animations.

al45tair, 2019-11-12:
To add to this, a couple of the preset animations appear to do this (I don't know if they all do)

elafargue, 2019-11-15:
I was able to flash one custom LED mapping + 'fingerprint' animation on the keyboard, but that was _very_ painful. Any further attempt resulted in a crashed keyboard firmware, either requiring a press on the reset button, or sometimes fn-F2 which seemed to work even if the keyboard looked dead.

Advice: if you manage to get one config that seems to work for you, save the JSON, you _should_ be able to go back to that config later on without having to redo everything...

al45tair, 2019-11-15:
@elafargue @xiletommy I know they've fixed something in the configurator back-end, so hopefully it's working a bit better than it was now. As far as the "crashed firmware" goes, a lot of the time I found it hadn't crashed, but the animations wouldn't work and the LEDs were all off. Sometimes you could get them going by pressing Fn-Shift-F6 or F7 (OTOH, I forget which; the one that's mapped to V-NEXT), other times not — but as long as the keyboard wasn't in Flash mode, it would still let you type.