kiibohd/configurator #156: Gemini Hexgears unresponsive and flickering after flashing static LED visualization

ryanzbone, 2019-11-13:
Keyboard does not work, as in typing does not produce output on my screen, after attempting to flash a static led visualization. The LEDs will flicker on and off as well. I'm able to recover functionality by hitting the flash button on the bottom of the keyboard, and then flashing the rainbow_wave animation as the startup animation, followed by unplugging it and plugging it back in. If I just flash it without unplugging the LEDs continue to flicker on and off.

noeljackson, 2019-12-09:
I ran into this issue too. What I found is this:

1. Create a custom "Animation" and set it as the default startup animation.
1. Flash the keyboard.
1. Unplug keyboard after success.
1. Using your mouse, quit the Hexgears Configurator
1. Plug keyboard back in.

It's a bug that needs to be fixed, but this is the workaround.

haata, 2020-03-24:
I've put in some fixes for this in the most recent firmware update. I'm not sure I've gotten it perfectly, but it should be much better than before.