kiibohd/configurator #159: LCD Screens Not Displaying Page and Layers Not Toggling

chamilitary-man, 2019-11-18:
Just flashed my keyboards and the LCD screens are no longer displaying which layer I am on. Also, if the layer is changed, it will not go back to any other layer.

Farreep, 2020-01-20:
Can confirm, I am having both of these issues.  LCD remains on white layer 1 logo, and layer changes are sticky/broken.  I can change into numpad layer, but not out of it.

tracer99, 2020-03-10:
I'm assuming this is for the Infinity ergodox. 

I was lucky in that I saved an old firmware from my layout and was able to re-flash it. But the new one is unusable.

cptbldbrd, 2020-03-13:
Yeah it was about the Infinity Ergodox. I bought a pair on r/mechmarket but after flashing it I ran into this issue. I have since gotten rid of them and built my own pair. It doesn't seem to like this issue is getting attention.

haata, 2020-03-14:
Please try the beta firmware release here https://github.com/kiibohd/controller/releases/tag/v0.5.7-beta1

https://github.com/kiibohd/controller/releases/download/untagged-482525da9bccdb2d4bad/ergodox.left.v0.5.7-beta1.dfu.bin
https://github.com/kiibohd/controller/releases/download/untagged-482525da9bccdb2d4bad/ergodox.right.v0.5.7-beta1.dfu.bin

If this helps, I can make it an official release and push it to the configurator.

tracer99, 2020-03-15:
@haata This firmware works great! Looks like all features are working properly including all layers switches and toggles.

Omnicraft, 2020-03-17:
@haata This definitely fixes the LCD issues on Ergodox infinity and needs to get updated so that the configurator can be usefull again. Manual compilation is nontrivial and most people dont' bother.

haata, 2020-03-17:
I'll be pushing it in the next couple of days (working through an annoying bug before I can push it to the configurator).

closed 2020-03-18 by cptbldbrd

tracer99, 2020-04-01:
Any ETA on this new release?

Omnicraft, 2020-04-01:
It's working for me on the latest configurator, I'm using an ergodox
infinity

tracer99, 2020-04-01:
N00b question. Is the configurator version independent from the firmware release? (as in does the configurator download it dynamically?)

haata, 2020-04-01:
Yes, the configurator version is independent. The firmware is built on a remote server and the configurator just downloads the binary to flash onto your keyboard.