kiibohd/configurator #160: No obvious way to update firmware for configurator builds

grizzlybus, 2019-12-12:
When I try to update my ergodox infinity via the Mac v1..0.2 kiibohd configurator, I get the following notification:

[image]

Yet- when I select my ergodox infinity, I don't have the option to download latest firmware like the various documentations on kiibohd configurator show:

[image]

If I simply flash, the resulting firmware does not seem to give me major functionality- layers can't be reverted, no LCD screen support. 

Please fix.

haata, 2020-03-24:
The firmware has been updated in the configurator, please try reflashing now.

uxjulia, 2020-04-02:
Experiencing the same issue. I see the notification that there's a firmware update, but no option to download. I have the latest version of the configurator installed.

huguesalary, 2020-04-08:
I am also running into this issue.

grizzlybus, 2020-04-08:
Click the flash button (make sure your desired key map is loaded!) and it
will flash to your keyboard once you set the keyboard in reset/flash mode

huguesalary, 2020-04-08:
I tried this a few times but unfortunately the compilation was failing silently, and as such nothing was happening.

I now have selected the default profile and flashed the keyboard with it and it seemed like it worked.

However, the interface makes this extremely confusing:

[image]

I assumed the green dot meant the new firmware was still to be installed. Now, I noticed it's missing the down arrow, so I imagine my keyboard was in fact correctly flashed to the latest firmware?

grizzlybus, 2020-04-08:
It sounds like it flashed.  Not sure what your keymap was, but for the one
that didn’t flash, did you validate it’s good JSON?

grizzlybus, 2020-04-08:
> I tried this a few times but unfortunately the compilation was failing silently, and as such nothing was happening.
>
> I now have selected the default profile and flashed the keyboard with it and it [...]

@huguesalary - the green dot means that a device of that keyboard type is detected... if the green dot has a downwards facing arrow, it means that there is a new firmware to download.  I verified by updating another keyboard today.

closed 2020-04-08 by grizzlybus