kiibohd/configurator #164: Hexgears Gemini Dawn/Dusk Firmware Update

ChristopherBui3, 2020-02-06:
For the Hexgears Gemini Dawn/Dusk, there have been major bugs regarding white LED hex code, and the Enter key LED does not work when assigning custom static color. There was a former issue regarding the latter bug, but I have tried the suggestions and it still does not work.

Per this [link](https://kiibohd.github.io/wiki/#/Keyboards/GeminiDawn?id=known-bugs) input club said there would be a firmware update. Does anyone if there has been one since the boards were shipped and where can we find these updates?

Thanks!

haata, 2020-03-14:
Looking into this (hopefully will make it with the upcoming firmware release).

haata, 2020-03-24:
New firmware has been pushed. Let me know if this fixes the Enter key issue (there is another place I can look to fix this issue).

akbyrd, 2020-05-25:
What about the LED color issue? And the configurator silently failing to flash if you edit/delete stock animations? And not being able to set the caps lock color based on state?

lolleko, 2020-09-26:
Any progress on this?

Configurator is basically unusable in the current state. Editing anything via the UI will prevent you from flashing.

ChristopherBui3, 2021-04-20:
Setting backlight LED to "#ffffff" still produces an orange-tint color. To date the only configurator compatible with the gemini dawn is found here: https://github.com/hexgears/configurator/releases

The other input-club keyboards have [updated firmware](https://github.com/hexgears/configurator) pushed, but it does not look like hexgears has not for the gemini/dusk keyboards. Any news on this would be appreciated.