kiibohd/configurator #171: No COM port exposed for ergodox infinity in latest configurator

BOTButtcheeks, 2020-03-29:
Why was this disabled?

last time I updated my keyboard was 2018 via the web interface and that still exposed the com ports

extra inconvenient since IED-LCD-Editor no longer runs meaning there is nothing to change the macro images in the firmware (it fails to do it on the base layer now, so now I have this rudimentary white I:C logo clashing with my keyboard's physical aesthetic which is beyond frustrating considering how much I have spent on customizing this keyboard, now there appears to be no solution to controlling the LCDs)

dylanmcauliffe, 2020-03-29:
If you compile the firmware manually, you can edit the LCD with [this info](https://www.reddit.com/r/MechanicalKeyboards/comments/5coiu8/guide_infinity_ergodox_linux_guide_altering_the/?utm_source=amp&utm_medium=&utm_content=post_body). That’s doesn’t address your issue directly, but is a workaround. I just used it to customize all my layers and the base image.

haata, 2020-03-29:
I'm transitioning to using HID-IO (instead of the virtual serial port) for CLI access. The main advantage here is that actual RPC is supported instead of just going through a CLI (that can be a bit buggy at times).

If you share with me the code (and OS) you're using I can try to rig up an example. Currently I can build examples in both Rust and Python.

While it's not really suitable for scripting (doesn't create a device but handles keystrokes directly right now), you can try out the HID-IO Core example if you're not afraid of compiling some rust (though you may need something recent, I usually use nightly).

https://github.com/hid-io/hid-io-core/blob/master/README.md#rpc-terminal-example

CrackedDilithium, 2020-04-01:
The code for the IED-LCD-Editor is located at https://github.com/chuckdee68/IEDLCDEditor.  I was trying to do the same thing as BOTButtcheeks and having the same issue.  I can't speak for him but I normally use Windows 10.

CrackedDilithium, 2020-04-01:
@dylanmcauliffe I just tried to go through that work around but I'm lost since it appears that the KLL files are no longer produced as part of the final output any more.  I tracked down in the log that the KLL files do get created during the compile process right after clicking "Flash Keyboard" but If someone has a suggestion on how to get those KLL files from the configurator I would be glad to try and take a swing at your suggested work around again.

dylanmcauliffe, 2020-04-02:
> @dylanmcauliffe I just tried to go through that work around but I'm lost since it appears that the KLL files are no longer produced as part of the final output any more. I tracked down in the log [...]

If you're trying to do my suggestion, I compiled my firmware using the instructions here: [https://kiibohd.github.io/wiki/#/Setup](https://kiibohd.github.io/wiki/#/Setup). The Docker method is easier when it comes to getting to/placing the files you have created from Windows. I created my .kll files by editing the defaults in the controller package.

BOTButtcheeks, 2024-05-10:
Hello,
I see your work on HID:IO is progressing. I don't see any evidence online of anyone modifying the LCDs on the ergodox since the change and I suspect the application of the usage is a bit distant in user ability. I attempted to compile HID:IO but unfortunately like most github projects I wasn't able to compile it with my limited knowledge (and seemingly programmer-lexia) and don't really have the time to figure it out. even if there was a pre-compiled application for windows or a compiled executable to pipe a bmp and a few commands to that would be fantastic. I know I'm essentially asking for free work for something I bought 8 years ago, but I gotta hope right?