kiibohd/configurator #172: Configurator Fails to Generate For Ergodox Infinity

thegreatco, 2020-04-01:
Trying to generate just the "blank" ergodox infinity keymap yields an error. Also, the Configurator does not surface this error to the user, I had to use the developer tools to even find it from the "Network" tab.

```
linux-gnu/Linux is supported.
Unix Build
Selected Generator: Ninja
/tmp/tmp.oyLMEUbztq/left
VID: 0x308F (Boot: 0x1C11)
PID: 0x0004 (Boot: 0xB007)
-- Build CPU Detected:
x86_64
-- Build Kernel Detected:
Linux 5.1.17-x86_64-linode128
-- Compiler Family:
arm
-- Compiler Selected:
gcc
-- Chip Selected:
mk20dx256vlh7
-- Chip Family:
mk2x
-- CPU Selected:
cortex-m4
-- Compiler Source Files:
Lib/kinetis.c;Lib/delay.c;Lib/entropy.c;Lib/gpio.c;Lib/periodic.c;Lib/storage.c;Lib/time.c;Lib/utf8.c;Lib/arm_cortex.c
-- Bootloader Type:
dfu
-- The C compiler identification is GNU 7.3.1
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Full Compiler:
/usr/bin/arm-none-eabi-gcc 7.3.1
-- Detected Scan Module Source Files:
Scan/Devices/ISSILed/i2c.c;Scan/Devices/ISSILed/led_scan.c;Scan/Devices/MatrixARMPeriodic/matrix_scan.c;Scan/Devices/STLcd/lcd_scan.c;Scan/Devices/UARTConnect/connect_scan.c;Scan/Infinity_Ergodox/scan_loop.c
-- Detected Macro Module Source Files:
Macro/PartialMap/kll.c;Macro/PartialMap/layer.c;Macro/PartialMap/macro.c;Macro/PartialMap/result.c;Macro/PartialMap/trigger.c;Macro/PixelMap/pixel.c;/tmp/tmp.oyLMEUbztq/left/generatedPixelmap.c
-- Detected Output Module Source Files:
Output/HID-IO/hidio_com.c;Output/Interface/output_gen.c;Output/USB/output_com.c;Output/USB/output_usb.c;Output/USB/arm/usb_desc.c;Output/USB/arm/usb_dev.c;Output/USB/arm/usb_keyboard.c;Output/USB/arm/usb_mem.c;Output/USB/arm/usb_mouse.c;Output/USB/arm/usb_rawio.c
-- Detected Debug Module Source Files:
Debug/cli/cli.c;Debug/latency/latency.c;Debug/led/led.c;Debug/print/print.c
-- Found Git: /usr/bin/git (found version "2.20.1") 
-- Found LSB_RELEASE: /usr/bin/lsb_release  
-- Build OS Detected:
Ubuntu 19.10 eoan
-- Found Ctags: /usr/bin/ctags-exuberant (found version "5.9~svn20110310") 
-- Found PythonInterp: python3 (found suitable version "3.7.5", minimum required is "3") 
-- Checking for kll
-- GITHUB_APIKEY was set.
-- kll Version Detected: kll 0.5.7.16
-- kll Installation Path: /root/.local/share/virtualenvs/Keyboards-41-guJFC/lib/python3.7/site-packages/kll
-- kll Layouts Cache Path: /tmp/10-hid-io-layouts-76840e4
-- Detected Layout Files:
/controller/Scan/Devices/ISSILed/capabilities.kll
/controller/Scan/Devices/MatrixARMPeriodic/capabilities.kll
/controller/Scan/Devices/STLcd/capabilities.kll
/controller/Scan/Devices/UARTConnect/capabilities.kll
/controller/Macro/PartialMap/capabilities.kll
/controller/Macro/PixelMap/capabilities.kll
/controller/Output/HID-IO/capabilities.kll
/controller/Output/USB/capabilities.kll
/controller/Debug/latency/capabilities.kll
/controller/Debug/led/capabilities.kll
/controller/Lib/arm_cortex.kll
/controller/Lib/kinetis.kll
/controller/Scan/Infinity_Ergodox/scancode_map.kll
/controller/Scan/Infinity_Ergodox/leftHand.kll
[... 158 lines cut ...]
[30/41] Building C object CMakeFiles/kiibohd.dir/Output/USB/output_usb.c.obj
[31/41] Building C object CMakeFiles/kiibohd.dir/Output/USB/arm/usb_mouse.c.obj
[32/41] Building C object CMakeFiles/kiibohd.dir/Output/USB/arm/usb_rawio.c.obj
[33/41] Building C object CMakeFiles/kiibohd.dir/Output/USB/arm/usb_dev.c.obj
[34/41] Building C object CMakeFiles/kiibohd.dir/Debug/latency/latency.c.obj
[35/41] Building C object CMakeFiles/kiibohd.dir/Debug/led/led.c.obj
[36/41] Building C object CMakeFiles/kiibohd.dir/Debug/print/print.c.obj
[37/41] Building C object CMakeFiles/kiibohd.dir/Debug/cli/cli.c.obj
[38/41] Building C object CMakeFiles/kiibohd.dir/Macro/PixelMap/pixel.c.obj
[39/41] Linking C executable kiibohd.elf
Setting new key
Key Prepended: 0000000000000000
[40/41] Message for DFU targets
 --> NOTE: kiibohd.secure.dfu.bin is required for secure Kiibohd Bootloaders.
     Only keyboards from late 2017 and onwards, support this.
     See lsusb -d 1c11: -v and look for iInterface 4 Kiibohd DFU Secure.
     Otherwise, use the kiibohd.dfu.bin file when it says iInterface 4 Kiibohd DFU.
     Use the physical reset button on the back of the keyboard to temporarily disable secure mode.
 --> NOTE2: Secure mode is disabled currently, but will be enabled when key negotiation is supported.
[41/41] Chip usage for mk20dx256vlh7
	 SRAM: 	43% 	28412/65536	bytes
	Flash: 	35% 	90852/253952	bytes
Firmware has been compiled into: '/tmp/tmp.oyLMEUbztq/left'
/controller/Keyboards
```

haata, 2020-04-01:
Try again, I've done some tweaking to the server.

thegreatco, 2020-04-01:
Hey @haata, thanks for the super quick response. Just flashed successfully! And now my screens are showing the layer number again too!

closed 2020-04-01 by thegreatco