kiibohd/configurator #173: NightFox Won't Flash Through Configurator

Xixeonn, 2020-05-06:
I just put together one the other day from someone who had bought this from the original release. When I was going to remap the keys I'm having troubles with the number keys (they are all f keys for some reason) and the backspace is delete and won't change through flashing. I followed the instructions from https://input.club/configurator-setup/ 
As well as the other site that was listed initially https://input.club/configurator-setup/
When trying to used Zadig I can’t find the recommended API Interface (Interface 5) To continue attempting through that method.
[image]
F1-12 is 1 through +
Esc is only flashing the device.
Backspace is delete
Page up and down are the volume up and downs.
Arrow keys are your page up and down plus delete and end.
I have flashed the device with Truefox profile many times.

haata, 2020-05-06:
Those instructions are quite out of date (thanks for reporting, totally forgot about that page!).

Please use this wiki now: https://kiibohd.github.io/wiki/#/Quickstart
Zadig is now integrated into the configurator so you don't need to worry about installing it.

Xixeonn, 2020-05-06:
I have tried that one as well sorry. It was meant to be that link and the other instead of a double post. No amount of flashing I've done has changed this so far.

closed 2020-05-06 by Xixeonn