kiibohd/configurator #177: Cannot compile and flash infinity 60% LED alphabet layout

cknv, 2020-06-22:
I backed the Infinity 60% LED quite a while ago on massdrop, but real life got in the way for a couple of years, and I am only now getting back to the keyboard. Note that the keyboard is an LED version using the Alphabet layout, although I haven't soldered the LEDs onto the board as I am not really that interested in lights on my keyboard.
The keyboard works in as-is, but I seem to be unable to make any changes to the layout.

I can load up the Alphabet layout just fine in the configurator, but when I press the "Flash Keyboard" button it spins for a split second and then stops. I cannot see any in-app logs (not even a button), but the network console reveals that the compilation failed, with a filename ending with `-error.zip` and a `success: false` bit in the JSON.

When I try the same with the AlphabetBlank layout, I am able to download a compiled firmware and probably (?) flash the keyboard, I can also confirm that the same pattern is repeating with Standard and StandardBlank as well has Hacker and HackerBlank.

It seems that problem is being triggered by using layers in the layout, as when adding a single mapping on Layer1 causes the compilation to fail. It seems to be isolated to this specific keyboard as compilation does not fail when mapping a key on layer1 on an otherwise blank whitefox aria layout.

I write "probably", as it seems that any changes I make have no effect: I tried changing the CapsLock key to Backspace (both via the UI and in the JSON itself), and I can confirm it is being sent to the server. However after flashing the key is unaffected and still functions as CapsLock.

For the record, I have tried both on a Win10 and an arch linux, both with the same result, although I haven't been as thorough replicating everything on the Win10 machine.

If there is any logs or result of commands I can provide to make this easier to figure out, just let me know, I'd be happy to help :)

rbn920, 2020-07-10:
I'm have the same problem. I was trying to flash to the default layout since I sold the keyboard. I am only able to flash the AlphabetBlank layout which is just the base layer. Any changes to that layout or attempts to flash the Alphabet layout do not work. Now I have a 60% keyboard with no layers...

LuisTellezV, 2020-08-13:
Can confirm as users above, only AlphabetBlank works, any other configuration fails, this only applies when selecting Infinity 60 LED as other options appear to be working.

I had the 1.02 configurator which is the version on the infinity guide and can confirm that when clicking flash keyboard as with cknv, button spins half a second and stops, I downloaded the latest version of the configurator on github which was version 1.1.0 and here it still doesn't work but instead of the half a second spin and stop, there is a red Compilation Failed message from Which i could see a log, which i pasted here **https://pastebin.pl/view/94a41c89** in case it helps, my theory from the little I could understand is that it seems to be a problem with the configurator failing to reach some files from your github repository.

If there is any other information I could provide in order to fix this problem just send me a message and I will provide as much information as a I can.

PS: For users above which are stuck with either a not functional keyboard or weird layout, as a temporary fix to keep the basics is to download the .dfu.bin file from the github repository for INFINITY 50 LED ALPHABET, it's a generic layout but is the only way to get all keys working AND still have LED working, you will only have base layer but can function until a more permanent solution works.

uxjulia, 2020-08-20:
I'm also having the same issue.

HaBeMe, 2020-10-28:
I have the same problem, trying to change anything with the current configurator is not possible (except the alhabetblank, which is utterly useless).

Is there any workaround for this? 

I saw some unanswered questions on Discord but no replies?

Has support for Infinity 60% and configurator stopped completely?

I managed to flash back an old configuration from \kiibohd-configurator\firmware-cache via de configurator using quick flash and selecting the .DFU. For now I atleast have a functioning keyboard again, but I would like to change some mappings.

cknv, 2020-10-30:
Yesterday I managed to flash a minimal colemak layout to my Infinity 60% keyboard which seems to have decent [support from QMK](https://github.com/qmk/qmk_firmware/tree/master/keyboards/infinity60).

I only tried the command line version using a `keymap.c` file, but I actually happen to like that approach as I can stick my config in a git repo and version the layouts.

So while that _is_ a workaround, it is also a fairly drastic departure as the Kiibohd Configurator is not used at all.

haata, 2021-03-02:
Sorry that this took so long. This should be fixed now.

Basically the filename extension needs to get truncated for the kll compiler, our rust server application was doing this a bit too gratuitously.

MD1.1-Standard-0.kll -> MD1 -> MD1.kll

No need to update your version of the configurator this is all on the server side.

closed 2021-03-02 by haata