kiibohd/configurator #189: Infinity Ergodox - Slave side not working

roygbyte, 2021-04-16:
I bought a 1.2b version of the Infinity Ergodox in 2017 via the Massdrop page. After updating the firmware using the Kiibohd configurator on Windows 10 I am experiencing an issue where the slave side of the keyboard does not appear to send keystrokes. 

I am using original cables from the drop. I've switched these cables around and tried different USB ports and different computers, including a Mac and my Ubuntu partition, to no avail. I've also reflashed some old firmware I compiled in 2018 from the online configurator. Same issue.

After doing a bit of looking around on forums, as well as this repo, it seems that this issue has occurred before. I've followed the steps mentioned in the threads but have not resolved the issue. 

I am not sure how to further troubleshoot this issue, or monitor log files to better diagnose what's going on. If someone could point me in the right direction to investigate this issue I'd gladly do the work. I'm just not sure what to try next!

OvahKeel, 2021-04-17:
You need to flash each side individually with the same firmware. Not sure if you tried that yet.

roygbyte, 2021-04-18:
Hi! Thanks for the suggestion. I tried that and the slave side still didn't work. Additionally, the output of the master was mirrored. From what I can gather from documentation, each side should be flashed separately with the firmware corresponding to that side  (i.e.: left.dfu.bin, right.dfu.bin).

I have a temporary solution that is letting me continue to type with my beloved keyboard... Plugging each side into a USB works (although if I'm on Linux, for whatever reason Xorg  doesn't play nice... so I have to use Wayland).

roygbyte, 2021-05-29:
Update on this issue.

This issue was ultimately resolved for me by replacing the USB cables, which hadn't been changed since the ~2017 Massdrop. [These USB 3.0 cables are confirmed to work](https://www.amazon.ca/gp/product/B01NGTUJ4Y/ref=ox_sc_act_title_1?smid=AREE309N4XPXI&psc=1). I should also mention that before switching cables I was able to "fix" the keyboard by installing QMK firmware. I say "fix" because the signal would still drop every now and then, *but*, at least it worked 90% of the time (compared to 0% of the time using Kiibohd). This could just be incidental (maybe the broken cable was correctly wiggled after I installed QMK... who knows!)

closed 2021-05-29 by roygbyte